DavidPagels/retro-pico-switch #1: Pico W Bluetooth

MrTechGadget, 2023-02-15:
Not an issue, but timing is perfect, they just added Bluetooth to the Pico W SDK a couple days ago, maybe this could be used to work with the 8BitDo or other Bluetooth switch controller adapters?

Cool project, look forward to making it

jars28, 2023-02-17:
>they just added Bluetooth to the Pico W SDK a couple days ago
> 

Welp. The Pico W on my desk has no other purpose now. Guess I should pick up another soon after I play around with getting a fight stick working.

DavidPagels, 2023-04-02:
Sorry for not responding sooner, I was hoping it wouldn't take this long, but I'm new to bluetooth and btstack needed some workarounds to get it working correctly with the switch. But I just got it working! It'll take a little bit longer to clean everything up and the first version will probably be a little rough while I add in some missing features.

closed 2023-04-02 by DavidPagels