DavidPagels/retro-pico-switch #3: Build guide

fortizc, 2023-02-27:
Hi, I have some problems to build the code and maybe it's a good idea write a small guide or improve the process. I tried to build but only I get a file called scr/retro_pico_switch.uf2 but that binary doesn't work on my rpi pico. How can I build the n64 binary? or the Gamcube binary?

MatthewTingum, 2023-06-10:
The USB and Bluetooth build variants are described in [CMakeLists.txt](https://github.com/DavidPagels/retro-pico-switch/blob/48393ae8486403806a2593328ac409c7e2f1b148/src/CMakeLists.txt#L3). Define `SWITCH_BLUETOOTH` for the Bluetooth variant. Undefine it for USB.

There are also prebuilt binaries on the [releases pages](https://github.com/DavidPagels/retro-pico-switch/releases).

Maybe this is a newer feature, but the controller (N64 / Gamecube) will be [automatically detected](https://github.com/DavidPagels/retro-pico-switch/blob/48393ae8486403806a2593328ac409c7e2f1b148/README.md?plain=1#L25).

MatthewTingum, 2023-06-11:
I think there could be a docker image used to build this though. That seems sane.

A discord channel for community support doesn't seem like a bad idea.

DavidPagels, 2024-11-01:
Local building is now available via a Dockerfile from PR #33

closed 2024-11-01 by DavidPagels