DavidPagels/retro-pico-switch #8: Problems Syncing with Blueretro Module

PavanTauh, 2023-06-16:
Trying to sync with Nintendo Switch and PC work fine, but trying to sync with bluetooth modules that require holding the sync button while turning on the controller do not work. For example, when trying to connect to a blueretro module such as this one (https://www.amazon.com/gp/product/B0BTLVLV4W/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1) I cannot get it to connect. The instructions on the module say to hold the sync button (which works on my other switch controllers) or to reprogram the bluetooth address using a computer for other controllers (such as the PS3). 

What would be the best way to connect this controller via bluetooth to the blueretro module if it asks to hold the sync button? Or is there a way to manually type in a bluetooth mac address?

TheIcedTeaAddict, 2023-12-31:
Hey, just got one built. The Pico seems to attempt to auto sync with a BlueRetro without a button when powered on, causing the BlueRetro device to stop blinking it's blue LED, however after a few seconds the Pico seemingly automatically disconnects, and the BlueRetro device resumes blinking. The Pico works with my switch and the BlueRetro works with other controllers, but not with each other unfortunately. Hopefully there will be an update to have it work.

Sbrivas, 2024-05-07:
Hi! Could you solve the compatibility between BlueRetro and this project?

> Hey, just got one built. The Pico seems to attempt to auto sync with a BlueRetro without a button when powered on, causing the BlueRetro device to stop blinking it's blue LED, however after a few [...]

Roet-Ivar, 2024-11-14:
what firmwares are you guys using? My picos just brick for any FW after the initial one.