DavidPagels/retro-pico-switch #10: Broken / ghost inputs in Windows - excluding Steam

lotuspar, 2023-07-02:
I've been trying out this project for a bit and I've been having a bit of an issue across multiple Pis (Pico W, Pico, USB-C Clone) and controllers (2 official controllers)

In the Windows game controller menu and Dolphin the controller appears to be freaking out (moving the analog stick or C stick does seem to make some sort of change - see the video)

In Steam Big Picture it seems to work fine, although with a tiny delay?

I've tried multiple 1k resistors (this result), no resistor (no result), the internal pull up resistor (notably also this result) 

Any help is appreciated.

https://github.com/DavidPagels/retro-pico-switch/assets/91263142/5d2c2eea-d8e1-4096-8552-87b5fb0ac000

lotuspar, 2023-07-02:
Tested on a separate Win 10 computer - same issue, but tested on Mac and seemingly works fine?

DavidPagels, 2023-07-15:
The descriptors the pro controller sends out aren't accurate, so windows doesn't know how to interact with the controller. Steam knows the correct descriptors and mostly ignores what the controller sends. The pro controller that this is emulating behaves the same.

closed 2023-07-15 by DavidPagels