DavidPagels/retro-pico-switch #11: No input but does show up in windows

QuentinWehkamp, 2023-07-14:
I can not for the life of me figure out why it is not working. it shows on windows as a controller but i cant get any input from it.

I have loaded it with the latest release of the usb.uf2 because i want to use it wired.
I have connected data -> gp18, 3.3v -> 3.3v, gnd -> gnd and a 1k resistor between gp18 and 3.3V. (i didnt wire up rumble because i dont want it)
I have tested continuity from the gamecube controller to the pins and that seems to work. 
I have measured the resistor to see if that was correct, it was.

I've been trying to build this ever since it was posted and i keep coming back to it but i can not figure out what i am doing wrong.

DavidPagels, 2023-07-15:
Sorry to hear it hasn't been working. Please post pictures of your current setup. Off hand, I'm not sure why it wouldn't be working unless maybe the two ground pins on the gamecube controller aren't internally connected and you're grounding the rumble ground instead of the data 3.3v ground.

RyanMac84, 2023-09-10:
I have the same issue using the usb.uf2 from the Rumble pack release, if I use the initial release, it works fine 🤷‍♂️

DavidPagels, 2023-09-10:
Same request as I had for the OP - Please post pictures of your current setup.

DavidPagels, 2023-09-10:
Whoops, didn't mean to close

RyanMac84, 2023-09-10:
[image]

Sakitoshi, 2023-09-16:
@RyanMac84 assuming red is connected to 3v3, black to ground and white to data in the controller, looks good.
as the adapter mimics a switch pro controller you need software for a pc to interpret it correctly. i've been using [this version of betterjoy](https://github.com/d3xMachina/BetterJoy/releases/tag/v8.0) that works on win 11 without issues and is updated to use hidhide instead of the outdated hidguardian, just make sure to change the "NonOriginalController" option to "defaultcalibration" or the stick and c-buttons may not work, alternatively you can just tick "AllowCalibration" for the same effect.

RyanMac84, 2023-09-25:
Thanks but I've just tried using Betterjoy and even though it shows as a Pro Controller when I 'Add Controllers' it doesn't show as Connected and the Locate button doesn't seem to do anything. As you can see in the built in Windows 11 Game Controllers, using the built in 'Test' is a complete mess. I don't get this this with the Initial release which works just fine. 
[image]

RyanMac84, 2023-09-25:
For reference, this is how the Initial release looks using Windows 11 Game Controllers  
[image]

Sakitoshi, 2023-09-25:
you also need to change the _NonOriginalController_ value to _defaultcalibration_ in betterjoy for the stick and c-buttons to work.
[image]
you can also install [hidhide](https://github.com/nefarius/HidHide/releases) and enable it (also shown in the picture above) to hide the unresponsive duplicate controller (4 axis 18 button joystick with hat switch) and prevent games from detecting it.
the Z button being detected as Z axis is normal, as the Z button is assigned as the L trigger of an xbox controller. for a more visual representation of how the system sees the controller use this web to test it: http://gamepad-tester.com.

as for MiSTer, i don't own one, but another person mentioned the same here https://github.com/DavidPagels/retro-pico-switch/issues/12#issuecomment-1731717222, you can look at the solution there.

closed 2024-11-01 by DavidPagels