DavidPagels/retro-pico-switch #12: c-buttons cancel each other

Sakitoshi, 2023-08-03:
Hi.
Don't know if this is how the NSO N64 controller also behaves on pc or not, i don't own a switch to test there, but only one c-button can be active at any given time, even if you press two c-buttons of different axes (c-down and c-left for example).
I'm using the usb version on a regular raspberry pico with an original n64 controller.

DavidPagels, 2023-08-03:
Since the project currently mimics a Switch Pro Controller, and that uses an analog stick for c button inputs, I have to translate c buttons to an analog stick output. I could've handle adjacent c-buttons, but what would I send if opposite buttons or 3+ c buttons are pressed?

Sakitoshi, 2023-08-03:
i managed to add the diagonals and found a solution for opposite directions by configuring c-up+down to be c-up and X and c-left+right to be c-right and Y.

first added the missing defines for the c-buttons in N64Controller.h:
```
#define N64_MASK_C_UP 0x8
#define N64_MASK_C_UPRIGHT 0x9
#define N64_MASK_C_RIGHT 0x1
#define N64_MASK_C_DOWNRIGHT 0x5
#define N64_MASK_C_DOWN 0x4
#define N64_MASK_C_DOWNLEFT 0x6
#define N64_MASK_C_LEFT 0x2
#define N64_MASK_C_UPLEFT 0xA
#define N64_MASK_C_UPDOWN 0xC
#define N64_MASK_C_LEFTRIGHT 0x3
```

then added the extra switches in N64Controller.cpp:
```
  switch (N64_MASK_C & _controllerState[1]) {
    case N64_MASK_C_UP:
      ry = SWITCH_JOYSTICK_MAX;
      break;
    case N64_MASK_C_DOWN:
      ry = SWITCH_JOYSTICK_MIN;
      break;
    case N64_MASK_C_LEFT:
      rx = SWITCH_JOYSTICK_MIN;
      break;
    case N64_MASK_C_RIGHT:
      rx = SWITCH_JOYSTICK_MAX;
      break;
    case N64_MASK_C_UPLEFT:
      rx = SWITCH_JOYSTICK_MIN;
      ry = SWITCH_JOYSTICK_MAX;
      break;
    case N64_MASK_C_DOWNLEFT:
      rx = SWITCH_JOYSTICK_MIN;
      ry = SWITCH_JOYSTICK_MIN;
      break;
    case N64_MASK_C_UPRIGHT:
      rx = SWITCH_JOYSTICK_MAX;
      ry = SWITCH_JOYSTICK_MAX;
      break;
    case N64_MASK_C_DOWNRIGHT:
      rx = SWITCH_JOYSTICK_MAX;
      ry = SWITCH_JOYSTICK_MIN;
      break;
    case N64_MASK_C_UPDOWN:
      ry = SWITCH_JOYSTICK_MAX;
      switchReport->buttons[0] = SWITCH_MASK_X;
      break;
    case N64_MASK_C_LEFTRIGHT:
      rx = SWITCH_JOYSTICK_MAX;
      switchReport->buttons[0] = SWITCH_MASK_Y;
      break;
```

like that, you have no conflicts and additionally you get access to the X and Y buttons.

closed 2023-08-07 by DavidPagels

BSeraph, 2023-09-22:
I'm having issues with the C-Buttons and this fix isn't included in the latest release. Would it be possible to provide a compiled build with this fix in it? Thanks in advance!

Sakitoshi, 2023-09-22:
@BSeraph i have it compiled on my repo https://github.com/Sakitoshi/retro-pico-switch/releases
i have been experimenting adding mario 64 all-stars compatibility and an analog stick curve better suited for gamecube style analog replacements. those hopefully will come soon™.

BSeraph, 2023-09-22:
Thank you so much! I'm testing this on MiSTer FPGA. The C-buttons are unuseable on it on the standard Right Stick mode, but they work perfectly via the Button Mode. Is there a way to make the Button mode the default mode, or does it have to reset everytime on boot?

Sakitoshi, 2023-09-22:
i don't know how to make it persistent yet, but the default mode can be easily changed by modifying https://github.com/Sakitoshi/retro-pico-switch/blob/70d720f1ee5af70bfdc9371a5e0e856761d2b68c/include/N64Controller.h#L25 to 1 and compiling.