DavidPagels/retro-pico-switch #16: can't get it to connect

annabellerinoa, 2023-08-15:
Dunno why but I can't get it to connect. The only change I have is putting the original rumble motor on pin 19 and editing the cone to match. Since I don't read c++I may have gotten the code wrong. If it is the code, can you tell me the right file and line to edit and how to edit it?

annabellerinoa, 2023-08-30:
OK. I got it to connect but still can't add the rumble feature to output to pin 19

DavidPagels, 2023-08-30:
Since you're motivated, this project may be a good opportunity to learn C++. It was my first language, so I know it's doable. Also, this isn't necessarily a good beginner project. You may want to start by working on some simpler embedded (pico/arduino/stm32/etc) projects first.

closed 2023-08-30 by DavidPagels