DavidPagels/retro-pico-switch #18: extra button mapping

annabellerinoa, 2023-08-30:
Needs a few more button maps:
 map the select button to like l+r+c up or something.
Zr+zl could be mapped to z+l+r 

This would give full functionality for the switch controls and allow the n64 controller to properly work with the gameboy, gba, snes, and nes nintendo online stuff. As it sits I can't save in zelda oracle of seasons without switching controllers and can't use the suspend and rewind controls

annabellerinoa, 2023-08-30:
Also, if snes x and y butting can be registered to respond to the c buttons, that would be nice.

DavidPagels, 2023-08-30:
For your first message - I don't want to make any additional button mappings without community consensus. You're free to make those changes on your own branch though. For your second message - with the latest code (by Sakitoshi) maps cup + cdown to X and cleft+cright to Y.

closed 2023-08-30 by DavidPagels