DavidPagels/retro-pico-switch #25: WaveBird compatibility?

Leumas-Eleets, 2024-05-03:
I've managed to get this mod running off of a couple AA batteries in a third-party N64 controller, and I thought that using a WaveBird controller would be the perfect solution to making it completely self contained. I've wired up the Pico W to a WaveBird controller and for the data line I have it connected to pin 7 of the wireless transmitter in the controller and I've connected VSYS to the middle pin on the power switch so that the Pico W is only powered on when the controller is powered on, and obviously I connected a ground pin on the Pico W to the controller's ground, but when I power it on the controller doesn't show up on my Switch nor does it show up in available Bluetooth connections on my phone. Does anyone know what pin on the WaveBird IC outputs the 24 bit signal needed for this mod, or if it's possible to modify this program to use the signal sent to the WaveBird transmitter that is used by the receiver to generate the 24 bit signal sent to the console?
[image]

DavidPagels, 2024-11-01:
It doesn't look like there's any trace that'd provide the signal required by this project within the wavebird itself. You'd have to grab it from the console-side transceiver.

closed 2024-11-01 by DavidPagels

CodeAsm, 2024-12-11:
Looks like that controller not only handles reading the button states, but also handles wirelessly transmitting this to the reciever.

you might be able to ditch this controller and read the buttons themselves (and save ton on wasted power) using https://gp2040-ce.info/

Which is exactly the oposite from what this very project does what I want, I want the pico to NOT skip decoding and do use the N64/GC connector.