DavidPagels/retro-pico-switch #29: Pico Zero compatibility

Roet-Ivar, 2024-09-13:
Hi. I've flashed the USB-version of you project onto a pico Zero using the tiny solder point for GP18, although nothing shows up on my pc after flashing  the board so I wonder if it's possible to make this compatible with a pico zero?

Thanks!

DavidPagels, 2024-09-13:
This project was coded specifically for the architecture of the RP2040 and won't work with a raspberry pi zero without a significant rewrite.

Roet-Ivar, 2024-09-13:
Hi. I think you might've miss read my question. I'm actually using a RP2040 Pico Zero (a tiny version of the rpi pico), but whenever I flash the usb-firmware it bricks the device.

Roet-Ivar, 2024-09-13:
> Ah, yep, I read too quickly. Did you also solder a pull up resistor to pin 18?

I did add a 1K resistor between data and 3.3v. Does this prevent it from being properly flashed you think? I always have to revert the firmware to stock to get it out of brick. Many thanks!

DavidPagels, 2024-09-13:
I wouldn't think so. It should just work. The pico zero is just an rp2040 on a different board.

Roet-Ivar, 2024-09-13:
> I wouldn't think so. It should just work. The pico zero is just an rp2040 on a different board.

I thought so too. I tried two pico zero boards just in case, this fw does not like them for some reason.

Roet-Ivar, 2024-09-13:
Wait. Your very first firmware "n64_pico_switch.uf2" actually showed up as a controller, the other ones doesn't. Hmm

Arakon, 2024-10-17:
While I can confirm that the Zero works with the USB firmware (note that it will NOT be recognized if no controller is plugged in), I have some really strange offsets.. i.e. the left stick in windows is constantly all the way to the left and only can be moved up and down (L/R seem to be mapped to the Z-Axis), and buttons 6 and 7 are rapidly triggering with no input. If I use P11etro's GCC-USB with the same Zero, it works just fine, so I'd say my hardware is good. I'm wondering if the GC/N64 detection is misfiring and it's thinking it's an N64 controller?

DavidPagels, 2024-11-01:
Resolved in #31

Roet-Ivar, 2024-11-04:
> I can confirm that it works on a Zero just fine. Using GP18, 1K resistor. Make sure you got your wiring right and that the grounds are both connected.

My issue is that the device no longer shows up in device manager after flashing, what fw version are you using?
I'm thinking it should still show up as a controller even if I haven't connected anything but the zero.