DavidPagels/retro-pico-switch #31: No input detected after flashing; flashing issues (Pico, wired N64 controller)

blockfeed, 2024-10-20:
I'm having a similar issue to #29. I was able to get the device flashed with the initial version ('n64_pico_switch.uf2'), and the device does show up on a PC as a POKKEN controller. However, no input is detected. I'm using a RPi Pico. 

When I flash with any of the newer firmwares, the device is no longer detected once flashed.

I have a 1K resistor between GP18 and 3.3v, and the wires are all connected to the N64 controller properly. I am testing using a Retro-bit 64 wired N64 controller. Please see the attached image.

[image]

DavidPagels, 2024-11-01:
Weird, it looks like the .uf2 on the release was corrupted somehow? I just tried flashing it myself and it didn't work. I flashed the exact same uf2 I still had locally and everything worked fine. I've now removed the old usb.uf2 and re-uploaded.

closed 2024-11-01 by DavidPagels

Roet-Ivar, 2024-11-01:
I tried the new usb upload but still have the same issue where it works on the first fw but not this last one.