DavidPagels/retro-pico-switch #32: Failing to Build

HerrTuring, 2024-10-26:
I have been trying to build this project in my own machine to try to make some changes. Tried both following Raspberry Pi Pico sdk examples that use cmake and simply using a docker image that is supposed to have everything related to the raspberry pi pico sdk and every time I try to build I always get:

#0 0.671 [  0%] Building CXX object src/switchController/CMakeFiles/switchBluetooth.dir/SwitchBluetooth.cpp.obj
#0 0.719 In file included from /app/include/SwitchCommon.h:4,
#0 0.719                  from /app/include/SwitchBluetooth.h:4,
#0 0.719                  from /app/src/switchController/SwitchBluetooth.cpp:4:
#0 0.719 /app/include/Controller.h:6:10: fatal error: hardware/pio.h: No such file or directory
#0 0.719     6 | #include "hardware/pio.h"
#0 0.719       |          ^~~~~~~~~~~~~~~~

Googling this hardware/pio.h error always gets me to people discussing arduino projects.

HerrTuring, 2024-10-27:
After some more googling I was able to build it. I am using a Dockerfile to build it without having to install anything in my machine. When I am done playing, I hope my pull request with a dockerfile and simple instructions for anyone wishing to build it themselves is welcome. :)

DavidPagels, 2024-11-01:
Resolved with PR #33 . Thank you for the contribution!

closed 2024-11-01 by DavidPagels