DavidPagels/retro-pico-switch #34: Still can't flash any of my picos

Roet-Ivar, 2024-11-14:
Hi, I know we've been discussing this before and those issues are now closed. I still can't flash any other release than the first one, is it A requirement that the controller is electrically connected in the later releases to even make it show up or should it show up as an input device anyway?

HerrTuring, 2024-11-15:
Not sure about the other releases, but I always have to have the controller connected before starting the pico as far as I know. Both tested with current release and built it myself. Both on RPi Pico W.

Probably due to the fact that in the code one of the things it checks while starting is which controller is connected. I assume that will not work if both GC and N64 are not connected.

zerofalcon, 2024-11-15:
> Not sure about the other releases, but I always have to have the controller connected before starting the pico as far as I know. Both tested with current release and built it myself. Both on RPi [...]

Hey friend! sorry to bother you. Can you please share your latest compiled binaries? ty in advance

HerrTuring, 2024-11-15:
> Hey friend! sorry to bother you. Can you please share your latest compiled binaries? ty in advance

My only contribution to this project was a quick guide on how to build it with docker. So, you can do it. xD
Anyway, for real, all the changes I made locally were a flop, so you could consider my last working build is basically the same as the latest release.

Roet-Ivar, 2024-11-28:
> My only contribution to this project was a quick guide on how to build it with docker. So, you can do it. xD Anyway, for real, all the changes I made locally were a flop, so you could consider my [...]

Maybe that's my problem, how I build the binaries. I going to try the docker way and see if makes any difference. I don't see why that adapter wouldn't show up at all if no controller was connected, I mean you should at least hear the USB connection sound.

HerrTuring, 2024-11-28:
> > > Hey friend! sorry to bother you. Can you please share your latest compiled binaries? ty in advance
> > 
> > 
> > My only contribution to this project was a quick guide on how to build it with docker. So, you can do it. xD Anyway, for real, all the changes I made locally were a flop, so you could consider my last working build is basically the same as the latest release.
> 
> Maybe that's my problem, how I build the binaries. I going to try the docker way and see if makes any difference. I don't see why that adapter wouldn't show up at all if no controller was connected, I mean you should at least hear the USB connection sound. Then again, you guys seem to have no problems using the officially released binaries.

Are you using a RPi Pico W? Because I had quite a headache to make it work in a RPi Pico 2 (2W just launched and I wasn't aware the 2 lacks the W's features). Anyway, all my tests are on the Switch, never actually tried it with PC.

Roet-Ivar, 2024-12-02:
I've tested the waveshare rp2040 zero, the regular original rpi 2040 pico w. Maybe I've got this project all wrong by thinking I can use this firmware for PC use. If you have a chance to test a working adapter I'd be very thankful!
In addition, I'd be interested to hear if plugged vs unplugged the controller makes a difference.
I appreciate your time!

Roet-Ivar, 2025-01-16:
@DavidPagels I've got myself another controller and it turns out that the adapter only shows up in windows if a controller is detected (turns out my other controller is faulty somehow). Maybe you could add this information in your instructions as a note for anyone struggling with at potentially bad controller. Thank you!

Roet-Ivar, 2025-01-16:
@DavidPagels I've got myself another controller and it turns out that the adapter only shows up in windows if a controller is detected (turns out my other controller is faulty somehow). Maybe you could add this information in your instructions as a note for anyone struggling with at potentially bad controller. 

I've tested the latest FW on the Switch and it works as expected, on Windows  you need to use steam big picture mode for correct controller inputs. Would it be hard to make it show up as an xinput controller if holding a button combo at startup?

I just want your comment on the latter before closing, thank you David!

closed 2025-01-26 by Roet-Ivar