DavidPagels/retro-pico-switch #36: Wiring diagram?

multigamesystem, 2024-11-26:
Could someone please provide a wiring diagram for wiring the GameCube controller connector to the Pico?

CodeAsm, 2024-12-11:
based on:
> The pre-built .uf2 file assumes the data pin is on the Pico's [GP18 pin](https://datasheets.raspberrypi.com/pico/Pico-R3-A4-Pinout.pdf). A 1k pullup resistor should also be wired between 3.3v and your data pin to work properly.
I havent build this yet, but for gamecube, this might be how to. i made this based on the above hint from https://github.com/DavidPagels/retro-pico-switch?tab=readme-ov-file#setup using https://wokwi.com and a picture editor
[
[image]
](url)

I hope you can figure out what pins of the connector go to where on the GC connector when you have one. if you snip it to wire directly, I have no idea what colors the cable have . https://simplecontrollers.com/blogs/resources/gamecube-protocol these numbers and layout I followed.
You could also move the resistor towards the pico connection wise (thus, solder from GP18 to 3.3v, and then attach the data wire of the GC)

CodeAsm, 2024-12-12:
Turns out my N64 cable was dead broken, bypassed it and this wiring diagram is confirmed for atleast the N64 and its 3 wires. so whatever is power pin, ground pin and data, if you can figure those out, this schema works

multigamesystem, 2025-05-11:
> Turns out my N64 cable was dead broken, bypassed it and this wiring diagram is confirmed for atleast the N64 and its 3 wires. so whatever is power pin, ground pin and data, if you can figure those out, this schema works

Thanks for your help I'm gonna attempt to wire up a test tomorrow.  I wonder if I'm able to get more than one controller input out of one Pico or do I have to use several to make a multi tap type input device?

multigamesystem, 2025-05-12:
Eh it sorta works but Im getting all sorts of random inputs when not touching the gamepad.  Like some sort of noise on the data.  It seems counterintuitive to connect the data pin to 3.3V?

multigamesystem, 2025-05-13:
> > Eh it sorta works but Im getting all sorts of random inputs when not touching the gamepad. Like some sort of noise on the data. It seems counterintuitive to connect the data pin to 3.3V?
> 
> **TLDR: use the resistor between data and 3.3v, do not connect data directly to 3.3v. I dont think the pico can read more than 1 controller at a time.**
> 
> You must connect trough a resistor (dont connect to 3.3v directly, you may fry the pico). the resistor is acting as a "pull up" resistor. if the code would support mutliple data lines for multiple devices, each dataline should get its own resistor to 3.3v. this allows each dataline to pull down the line to ground. (its often easier for microcontrollers to pull to ground, then to pull to supplyvoltage) https://en.wikipedia.org/wiki/Pull-up_resistor

Have you actually wired up a single Official GameCube controller using this PICO program and wiring diagram?  Could you test?  I cant get it to work.

multigamesystem, 2025-05-14:
Ive designed a casing for the USB-C version of the Pico.  I already did a test print with my resin 3D printer.  It came out good.  I also just designed a little PCB that solders to the end of the GameCube port jack to make soldering wires and the resistor easier.  I want to make sure the wiring all makes sense before I commit to having these little PCBs made.

[image]
[image]

multigamesystem, 2025-05-14:
Also designed a 4 port version with USB hub inside to handle the 4 inputs.  While this seems wasteful and maybe expensive to achieve something that's already being done with a single soc and pcb.  I could make my own RP2040 board but thats maybe beyond my PCB skillset.  

[image]
[image]
[image]

multigamesystem, 2025-05-14:
Details of my GC Port add on pcb.  I hope it makes sense how the wiring is setup.  I used your guide as reference.  

[image]
[image]