DavidPagels/retro-pico-switch #41: Change data pin to something else?

Jaxel, 2025-04-28:
Would it be possible to change the data pin from GPIO-18 to somewhere else? I want to use this on the much smaller RP2040zero, but it doesn't have easy access to GPIO-18.

[image]

Jaxel, 2025-04-28:
I figured out how to do it. In `main.cpp`, there is a variable for the pin assignment. I just changed it to 14.

multigamesystem, 2025-05-14:
> I figured out how to do it. In `main.cpp`, there is a variable for the pin assignment. I just changed it to 14.

Could you share the .UF2 with this changed data pin?

Borrajo, 2025-06-29:
I modified the uf2 to connect it with the pin 15. it's works

[usb_pin15.zip](https://github.com/user-attachments/files/20970370/usb_pin15.zip)